meekrosoft/fff #9: EXTERN_C clashes with macro of same name in winnt.h

CiderMan, 2016-04-08:
We have a few files in our unit-tests that need fff.h but also need to include windows.h. Unfortunately, both define a macro EXTERN_C so we get warnings about macro redefinitions:

C:\Program Files (x86)\Microsoft Visual Studio 8\VC\PlatformSDK\include\winnt.h(359) : warning C4005: 'EXTERN_C' : macro redefinition
        W:\dyn_view\common\interface\unittest/fff.h(86) : see previous definition of 'EXTERN_C'

I suggest that the EXTERN_C/END_EXTERN_C in fff.h be prefixed with FFF_ to avoid this clash. (This is the change we've made locally with search-and-replace and it has resolved the issue.)

Regards,

Steve.

meekrosoft, 2016-04-08:
This is a great idea!  Could you send me a pull request an I will integrate it in fff

CiderMan, 2016-04-08:
Thanks Mike. Since this is an issue at work, I'm just confirming that I can assign IP/copyright for this change before I start. (Currently I've only patched fff.h, not the generation script.)
I assume that I just need to update the .rb file, install Ruby and confirm that it produces a usable fff.h?

CiderMan, 2016-04-08:
You have a pull request!

meekrosoft, 2016-05-16:
Great, thanks so much!

closed 2016-09-23 by CiderMan