meekrosoft/fff #13: Have the option to make the fakes static functions

frankbenoit, 2016-06-29:
Hi

to have the testing code in different modules and not having the need to think about unique names, it would be good to have a way to make the fakes static function.

frankbenoit, 2016-06-29:
Or the option to use the separated DECLARE_ and DEFINE_ macros.
Currently the DECLARE_ macro is missing the functions prototype. So it is not possible to but the fakes into a separated modules and have them globally available.
E.g.:

#define DECLARE_FAKE_VOID_FUNC1(FUNCNAME, ARG0_TYPE) \
    FFF_EXTERN_C \
        typedef struct FUNCNAME##_Fake { \
            DECLARE_ARG(ARG0_TYPE, 0, FUNCNAME) \
            DECLARE_ALL_FUNC_COMMON \
            void(*custom_fake)(ARG0_TYPE arg0); \
        } FUNCNAME##_Fake;\
        extern FUNCNAME##_Fake FUNCNAME##_fake;\
        void FUNCNAME##_reset(); \
        void FUNCNAME(ARG0_TYPE arg0); \    // <-- this line is needed
    FFF_END_EXTERN_C \

And this in a way, so it can also detect the arguments automatically. Because now, i can only use the makros with the explicit argument number in the name.

usr42, 2017-01-24:
Hi Frank,

can you please check, if my fix is working for you?

Cheers,
Balthasar

closed 2018-01-29 by meekrosoft