meekrosoft/fff #21: (Feature request) Change call_history type to a function pointer

dmcminniwc, 2017-01-23:
Could fff be changed so that instead of `fff_globals_t.call_history` being an array of void pointers it is an array of function pointers? The reason for asking is that I am using fff in an embedded project which uses IAR for msp430, although I think you'd come across this with other compilers/architectures too. The particular processor architecture I'm using can address 20 bits but the data pointer size varies (16/16 or 32 bits) with the model selected (Small/Medium or Large respectively) in the project options whereas the function pointer size is always 32 bits. The function pointers in the call history get truncated in Small/Medium models because `void *` is treated as a data pointer and therefore its size changes with the model.

Changing from `void *call_history[FFF_CALL_HISTORY_LEN];` to something like:

    typedef void (*fft_function_t)(void);

    typedef struct { 
        fff_function_t call_history[FFF_CALL_HISTORY_LEN];

would allow this to work correctly for this scenario and maintain compliance with C99 (section 6.3.2.3, paragraph 8 - I assume also other versions of the standard but haven't checked). The user would then cast their mock functions to (fff_function_t) when comparing to the call history elements.

usr42, 2017-01-23:
Hi dmcminniwc,

I created a branch to address your issue:
https://github.com/usr42/fff/tree/issue21

Can you check if this is working for you? If yes I'll open a pull request to merge it to meekrosoft's repository.

Best regards,
Balthasar

dmcminniwc, 2017-01-23:
Yes, that is working for me.

usr42, 2017-01-24:
My pull request to fix for issue 13 is not merged yet. So I created a branch fixing your issue (13) without the changes to fix issue 13: https://github.com/usr42/fff/tree/issue21_without_issue13

I'll open a pull request for the cleaned branch.

meekrosoft, 2017-02-03:
Thanks for the suggestion @dmcminniwc - this was an interesting use case.

+ thanks @usr42 for the implementation.  This has been merged in!

closed 2017-02-03 by meekrosoft