meekrosoft/fff #26: Faking functions that take no parameters should use void in parameter list

dmcminniwc, 2017-01-26:
If I use any of the FAKE_*_FUNC0 variants then fff declares functions which take any number of parameters because it uses an empty parameter list. For example, if there is a function `foo` which has no arguments and returns nothing then using `FAKE_VOID_FUNC0(foo);` eventually ends up with a function declaration:

    void foo();

If I have an existing header for the code under test that declares foo as:

    void foo(void);

because, really foo doesn't have any arguments and I want to ensure the compiler catches me trying to pass the wrong number of arguments instead of just accepting any and (possibly) leading to subtle runtime bugs. However that is then in conflict with the declaration from fff and you get compiler warnings (IAR by default, MSVC must enable all warnings).

Would it be possible to use void instead of an empty parameter list? This would apply to all the other no-argument functions such as the custom delegate, custom delegate sequence and reset functions.

usr42, 2017-01-26:
That is a good finding. Thanks! 
I'm going to fix that and open a pull request.

usr42, 2017-01-26:
Can you please check my pull request, if it is working as intended?
https://github.com/meekrosoft/fff/pull/27

meekrosoft, 2017-02-03:
Another great one - thanks both!

closed 2017-02-03 by meekrosoft