meekrosoft/fff #29: Add documentation and tests for VARARG fakes

usr42, 2017-02-06:
FFF also supports faking VARARG functions, e.g. 
`void voidfunc3var(const char *fmt, int argc, ...);` 
could be faked with 
`FAKE_VOID_FUNC_VARARG(voidfunc3var, char *, int, ...);`

But how to use this is not described in the readme. 
Additionally the unit tests for VARARGs are not sufficient. There are no assertions. I think there should be at least be a test for argument capturing. Also the custom_fake for VARARGs should be covered by unit tests.

rubiot, 2017-02-06:
I just submitted a pull request adding a section in the README explaining how to fake variadic functions. Please check it.

usr42, 2017-02-06:
Thank you! I like the explanation with printf very much.

usr42, 2017-02-06:
@rubiot I just improved the vararg tests. I'll open a pull request to your branch.

usr42, 2017-02-06:
The pull request is open: https://github.com/rubiot/fff/pull/1

rubiot, 2017-02-06:
Very nice!