meekrosoft/fff #33: Dealing with reference/pointer arguments?

martijnthe, 2017-05-08:
What's the best way to mock functions that take references/pointer arguments that usually point to temporary (stack allocated) variables? If you have a lot of this in your code, making custom fakes for everything lowers the value of using FFF in the first place...

Can you see a way to support dereferencing pointer arguments and copying the actual values to the history? It would have to be on a per-argument basis because often you also don't want to deref the pointer (i.e. an out-pointer).

meekrosoft, 2018-03-07:
Hi Martijn,
Sounds like an interesting use case.  Do you have an implementation in mind?  I'd love to take a look.
Cheers,
Mike

mayl, 2018-03-29:
I don't have an implementation in mind, but just want to chime in that I too have a use case for this.  I have several functions who's purpose is to generate a buffer on the stack and then pass that buffer to a second function.  I would like to fake the second function and test for proper content in these buffers, but as @martijnthe points out the only good way to do this currently seems to be a custom fake.

BillDenton, 2022-07-29:
Any updates? I have:

1. Functions that take pointers where the structure that it points to needs to be stored for checking later. I can do this via a custom fake. I declare a fake with pointer parameters changed to the structure. Then I can that from a fake function. Easy to do but would be better if there was a way of doing this.
2. Functions that references. I use the similar approach. Obviously you can't make arrays of references.

rdlaner, 2022-09-30:
+1 to this. Having to create a custom fake just to copy out the value pointed to by a pointer argument is not ideal. It'd be really great if the arg history contained copies of the dereferenced pointer args.