meekrosoft/fff #34: Readme describes nested function example as standard conforming

gsauthof, 2017-05-11:
This statement from the README isn't accurate:

> In case your project uses a C99 compliant C compiler you can even combine all this in a single unit test function so you can easily oversee all details of the test.

in the context of the code example that follows the paragraph:

    // ...
    TEST_F(FFFTestSuite, when_value_custom_fake_called_THEN_it_returns_custom_output)
    {
        Time t;
        Time getTime_custom_now;
        void getTime_custom_fake(Time *now) {
            *now = getTime_custom_now;
        }
       // ...
    }

ISO C99 doesn't specify nested functions. Nested functions aren't included in C11, as well. They are a GCC extension.

Perhaps you want to change this sentence to something like: _In case your project uses a C compiler that supports nested functions (e.g. GCC) you can even combine all this in a single unit test function so you can easily oversee all details of the test._

meekrosoft, 2017-12-18:
Thanks for the clarification @gsauthof !

closed 2017-12-18 by meekrosoft

jasen-b, 2018-06-11:
well, it's only recently that microsoft produced a C99 compiler that doesn't support nested functions (prevously MSVC had been C89)

commit 82eab516 (2017-12-18): Update to readme to correct misinformation on C standard

It turns out that nested functions are a gcc thing - not standard C.

Closes #34