meekrosoft/fff #36: cmake - Multiple definition error

avk2, 2017-09-08:
This is how I defined my function:
```
#include "lib/fff.h"
DEFINE_FFF_GLOBALS;
FAKE_VALUE_FUNC(bool_t, probe_is_ok, adc_probe_t);
/* List of fakes used by this unit tester */
#define FFF_FAKES_LIST(FAKE)            \
  FAKE(probe_is_ok)                    \
```

And this is what I get when I build

```
../app/middleware/libmiddleware.a(probe.c.o): In function `probe_is_ok':
~/AppProject/app/middleware/probe.c:19: multiple definition of `probe_is_ok'
CMakeFiles/unit_tests.dir/AppTest.cpp.o:~/AppProject/test/AppTestTest.cpp:16: first defined here
```

dturner, 2017-09-17:
Also running into this. My exact linker error message is: 

`ld: 1 duplicate symbol for architecture i386`

dturner, 2017-09-17:
Think I figured it out. In my case at least, I was linking against the object which I was mocking, as well as the fake. 

It looks like you're doing the same thing. Make sure `probe.c.o`is excluded from your unit test binary.

meekrosoft, 2017-09-17:
Hi folks,
Yes, the fake will clash with an existing function definition if it is linked in...be sure not to include it in the test binary.
I will close this issue...

closed 2017-09-17 by meekrosoft

avk2, 2017-09-18:
If I dont link it, how is supposed I make it work?

program to test --> depends on some external programs (linked)
test --> depends on program to test (linked)

```
target_link_libraries(unit_tests gtest gtest_main)
target_link_libraries(unit_tests app)
```

meekrosoft, 2017-09-18:
The trick is to put functions that are faked in tests in separate translation units (object files) that are only linked in production.  An uglier alternative, is to use the preprocessor to `#ifdef PROD` away the function during test builds

ekeroc, 2018-08-07:
So, that means the fake function and original function can't be called in the same object file?
In other words, if I can't call or use the original A function after I fake the A function?

meekrosoft, 2018-08-08:
Yes, you are correct.   The whole point of fff is to create a replacement function for testing a translation unit without changing the production code.

If the code you want to test calls something you want to assert on, e.g.

`````
void do_something(void){
  counter++;
  another_function(counter);
}
````

You want to test this code works by asserting that `another_function` is called, and with the correct parameters....without changing the production code.  The only way to do so is with the preprocessor or the linker.

wulfgarpro, 2018-11-23:
Response #55.