meekrosoft/fff #37: Feature Request for expectations

stefanpoeter, 2017-09-23:
Hi all,

imagine the following scenario:

```
FAKE_VOID_FUNC(__wrap_write, int, void*, ssize_t)

SOME_TEST_FUNC() {

    struct some_struct send = { ... };

    method_that_calls_write(args, that, create, a, object, like, send, on, the, stack);

    assert_mem_eq(&send, __wrap_write_fake.arg0_value, sizeof(send));

}
```

The `method_that_calls_write` create a object on the stack that look like `send` and writes its content. After the call to `method_that_calls_write` the created object on the stack is not accesible anymore, hence, I cannot check the written data.

Is there a feature planned to make expectation to a mock object and check these expectations later on?

meekrosoft, 2017-09-25:
Hi Stefan,

I'm not sure I 100% understand what you need, but could it be solved with a custom return value delegate for `write` that captures what is passed?
https://github.com/meekrosoft/fff#custom-return-value-delegate

There is an example test that shows how it is done:
https://github.com/meekrosoft/fff/blob/4e7451acd15715c95b275ebab6e77d9eb5b9cd7b/test/test_cases.include#L215

Best,
Mike

stefanpoeter, 2017-09-25:
Hi Mike,

that could work, thanks, I'll try it later.

Greetings,
Stefan

stefanpoeter, 2017-10-08:
https://github.com/meekrosoft/fff#how-do-i-fake-a-function-that-returns-a-value-by-reference

Thats what I was looking for. Thanks for your help.

closed 2017-10-08 by stefanpoeter