meekrosoft/fff #41: Function Overloading Support

1q2w3e4rfv, 2018-01-25:
I have two functions with the same name but different parameters:
Set-up for multiple test-suites.

`int tri1021_set_axis_param(const StepperMotor_t*, enum T1021_AxisParameters, uint32_t);`
`int tri1021_get_axis_param(const StepperMotor_t*, enum T1021_AxisParameters);`

public header
`DECLARE_FAKE_VALUE_FUNC0(int, tri1021_set_axis_param, StepperMotor_t*, enum T1021_AxisParameters, uint32_t);`
`DECLARE_FAKE_VALUE_FUNC1(int, tri1021_get_axis_param, StepperMotor_t*, enum T1021_AxisParameters);`

private c file
`DEFINE_FAKE_VALUE_FUNC0(int, tri1021_set_axis_param, StepperMotor_t*, T1021_AxisParameters, uint32_t);`
`DEFINE_FAKE_VALUE_FUNC1(int, tri1021_get_axis_param, StepperMotor_t*, T1021_AxisParameters);`

Has compiler error: 
cannot determine which instance of overloaded function "tri1021_set_axis_param" is intended

Is it possible to support overloading?

meekrosoft, 2018-01-28:
Mmmmm....
I guess that a way to do it would be to append the number of arguments to the struct name.

1q2w3e4rfv, 2018-01-29:
I have a work around, I set one function to have a const StepperMotor_t*.