meekrosoft/fff #45: old version fff to latest transition trouble due to extern "C"

tlawork, 2018-03-06:
I work at Cisco and we took a TDD class from James Grenning a while back and he introduced us to FFF. That was many years ago and we've been using an old version.  We need to move to the latest because of our need for the ____VARARG fakes.  However, in looking at the changes, there has been a lot of advancement since we pulled our copy - most notably there is a lot of changes regarding extern "C" declarations throughout dependent on #ifdef __cplusplus.  

In particular, this is causing most of my existing unit tests to fail - probably because we took matters into our hands to address the placement of extern "C" - so as example, I have a unit under test in a C file that calls getpid.  The test program is a CPP file and that is where FFF is included and getpid is faked.

On the call to ```FAKE_VALUE_FUNC(pid_t, getpid)``` it fails because of inconsistency:

```
In function 'pid_t getpid()': declaration of 'pid_t getpid()' with C language linkage
conflicts with previous declaration 'pid_t getpid()' due to different exception specifications
```

So my issue is - can you summarize what has changed in regard to extern C and how to handle that since the various upgrades through the years?

Should I be using extern C at all in my CPP test programs?

TODDL
Cisco Systems.

tlawork, 2018-03-06:
Here is more information. I added -E to the g++ compile and looked at the preprocessed output. Searching on getpid - I found the following is the FFF fake for getpid - and sure enough it is wrapped with extern "C" - yet I am running this on a CPP compiler and I made sure __cplusplus is set.  So why is the fake for getpid wrapped with extern C????  or is that the design that all fakes are based on C? If so how do I make sure that the prototype declaration for the item (in this case getpid from unistd.h) is C-ized?

```
extern "C"{ typedef struct getpid_Fake { unsigned int call_count; unsigned int arg_history_len; unsigned int arg_histories_dropped; pid_t return_val; int return_val_seq_len; int return_val_seq_idx; pid_t * return_val_seq; pid_t return_val_history[(50u)]; int custom_fake_seq_len; int custom_fake_seq_idx; pid_t(*custom_fake)(void); pid_t(**custom_fake_seq)(void); } getpid_Fake; extern getpid_Fake getpid_fake; void getpid_reset(void); pid_t getpid(void); } extern "C"{ getpid_Fake getpid_fake; pid_t getpid(void){ if(getpid_fake.call_count < (50u)){ } else{ getpid_fake.arg_histories_dropped++; } getpid_fake.call_count++; if(fff.call_history_idx < (50u)) fff.call_history[fff.call_history_idx++] = (fff_function_t)getpid;; if (getpid_fake.custom_fake_seq_len){ if (getpid_fake.custom_fake_seq_idx < getpid_fake.custom_fake_seq_len){ pid_t ret = getpid_fake.custom_fake_seq[getpid_fake.custom_fake_seq_idx++](); if ((getpid_fake.call_count - 1) < (50u)) memcpy((void *)&getpid_fake.return_val_history[getpid_fake.call_count - 1], (const void *) &ret, sizeof(ret));; return ret; } else{ pid_t ret = getpid_fake.custom_fake_seq[getpid_fake.custom_fake_seq_len-1](); if ((getpid_fake.call_count - 1) < (50u)) memcpy((void *)&getpid_fake.return_val_history[getpid_fake.call_count - 1], (const void *) &ret, sizeof(ret));; return ret; return getpid_fake.custom_fake_seq[getpid_fake.custom_fake_seq_len-1](); } } if (getpid_fake.custom_fake) return getpid_fake.custom_fake(); if (getpid_fake.return_val_seq_len){ if(getpid_fake.return_val_seq_idx < getpid_fake.return_val_seq_len) { if ((getpid_fake.call_count - 1) < (50u)) memcpy((void *)&getpid_fake.return_val_history[getpid_fake.call_count - 1], (const void *) &getpid_fake.return_val_seq[getpid_fake.return_val_seq_idx], sizeof(getpid_fake.return_val_seq[getpid_fake.return_val_seq_idx])); return getpid_fake.return_val_seq[getpid_fake.return_val_seq_idx++]; } if ((getpid_fake.call_count - 1) < (50u)) memcpy((void *)&getpid_fake.return_val_history[getpid_fake.call_count - 1], (const void *) &getpid_fake.return_val_seq[getpid_fake.return_val_seq_len-1], sizeof(getpid_fake.return_val_seq[getpid_fake.return_val_seq_len-1])); return getpid_fake.return_val_seq[getpid_fake.return_val_seq_len-1]; } if ((getpid_fake.call_count - 1) < (50u)) memcpy((void *)&getpid_fake.return_val_history[getpid_fake.call_count - 1], (const void *) &getpid_fake.return_val, sizeof(getpid_fake.return_val)); return getpid_fake.return_val; } void getpid_reset(void){ memset(&getpid_fake, 0, sizeof(getpid_fake)); getpid_fake.arg_history_len = (50u); } }
```

meekrosoft, 2018-03-07:
Hi Todd,
Could you send me a cut down way to reproduce so I can investigate?
Thanks,
Mike

tlawork, 2018-03-07:
Sorry for the size of this - but in order to make this work for you without a lot of trouble - I precompiled, google test and the objects are left in this zip.  
[ciscoduplication.tgz.zip](https://github.com/meekrosoft/fff/files/1790671/ciscoduplication.tgz.zip)

Unzip then CD to stub.
run the makefile and it should cd to _build folder and run the c++ command that fails.

meekrosoft, 2018-03-14:
Hi @tlawork - Sorry for the slow response, I'm traveling a lot these days...
Can you confirm that you can use the latest `fff.h`?  Or by different direction, do you mean you have found a different testing solution?
Thanks!

tlawork, 2018-03-14:
by using the latest fff.h I get an error because the C++ test routine (specifically gtest)  includes unistd.h as a c++ module, but FFF is faking it as a C module and the linker complains.  We are trying to fake out the function getpid() which is defined in unistd.h

By different direction - we are dropping our meson/ninja investigation and it was changing the build system that prompted this error - we do not get this error when using the older version of FFF so we are simply going to stick with the older version for now - so we are still on board using FFF - its very effective.

TODDL (Cisco Systems)

wulfgarpro, 2018-12-14:
@tlawork please try the latest version of fff.  A lot of the C extern mess has been removed.   I managed to fake `getpid` like so:

#include <unistd.h>
#include <stdio.h>
#include <assert.h>

#include "fff.h"

DEFINE_FFF_GLOBALS;
FAKE_VALUE_FUNC(pid_t, getpid);

pid_t custom_getpid()
{
    return 0xdeadbeef;
}

int main(void) {
    getpid_fake.custom_fake = custom_getpid;
    pid_t pid = getpid();
    assert(getpid_fake.call_count == 1);
    assert(pid == 0xdeadbeef);
    return 0;
}

closed 2018-12-14 by wulfgarpro