meekrosoft/fff #46: Issue with functions with void pointers as arguments

Flockstud, 2018-03-29:
Hi,

I'm having issues trying to mock a function that has void pointers. It seems to be calculating the size of the argument to save to array, but this would probably cause an issue. 
Is it possible to mock this kind of function? I guess we'd have to use a different function than memcpy()?

This is an example:
unsigned int SelfLib_Write(void *var_1, void *var_2, unsigned char b);

Typical error with gcc:

> error: too many arguments to function '*(SelfLib_Write_fake.custom_fake_seq + ((sizetype)((long long unsigned int)SelfLib_Write_fake.custom_fake_seq_len * 8ull) + 18446744073709551608u))'
 DEFINE_FAKE_VALUE_FUNC(UI_32, SelfLib_Write, void *, void *, UI_32);

Is this library still maintained?

Thanks

wulfgarpro, 2018-11-23:
@Flockstud this example works for me without a hitch:

```c
#include "fff.h"

DEFINE_FFF_GLOBALS;

FAKE_VALUE_FUNC(unsigned int, SelfLib_Write, void*, void*, char);
unsigned int custom_SelfLib_Write(void*, void*, char)
{
    return 0x1;
}

int main(void)
{
    SelfLib_Write.custom_fake = custom_SelfLib_Write;
    return 0;
}
```

Are you still having a problem?

closed 2018-11-25 by wulfgarpro

Flockstud, 2018-11-26:
Hi wulfgarpro!

Thanks for your reply, sorry for long wait, a lot has happened.
It looks like that would work! 
Thanks and i hope to get it back working! 

Best Regards,

Joel