meekrosoft/fff #51: Attempting to FFF FreeRTOS.. getting "error: use of deleted function ... "

williamhuang03, 2018-06-18:
Hello,

I am trying to stub out some FreeRTOS API using FFF. At the moment, I am trying to stub out xTimerCreate:

`FAKE_VALUE_FUNC(void *, xTimerCreate, const char * const, const TickType_t, const UBaseType_t, void * const, TimerCallbackFunction_t);`

However, I am getting a compilation error: 
```
error: use of deleted function 'xTimerCreate_Fake::xTimerCreate_Fake()'
     FAKE_VALUE_FUNC(void *, xTimerCreate, const char * const, const TickType_t, const UBaseType_t, void * const, TimerCallbackFunction_t);
```

Unfortunately, I cannot change the production code to abstract out FreeRTOS.

Any ideas how I can get around this?

Thanks

williamhuang03, 2018-06-18:
This was the same as https://github.com/meekrosoft/fff/issues/3

I noticed the fix was for C only, so I've compiled the file under C-only and it seems to be compiling that file with no errors.

Out of curiosity, I am wondering why is the fix C-Only? i.e. what limitations of the fix does it not translate to C++ as well?

meekrosoft, 2018-06-22:
Hi William,
Glad for the feedback.  I suppose the fix in C-only since that was what was in the pull request.  I would be happy to accept a pull request that adds C++ support :-)
Best regards,
Mike

closed 2018-06-22 by meekrosoft

leonardopsantos, 2025-01-14:
While a C++ fix is not forthcoming, there's a possible [ugly workaround](https://github.com/meekrosoft/fff/pull/85#issuecomment-2588484246) that I've been using.