meekrosoft/fff #53: GCC warning for variadic with no arguments

cshorler, 2018-07-22:
I'm mocking several functions with declarations similar to this:

`void SCHEMAinitialize();`

using:
`FAKE_VOID_FUNC(SCHEMAinitialize)`

currently results in a compiler warning:

    /home/chorler/projects/src/stepcode/src/express/test/test_express.c:56:32: warning: ISO C99 requires at least one argument for the "..." in a variadic macro
     FAKE_VOID_FUNC(SCHEMAinitialize)
                                    ^
current workaround:
`FAKE_VOID_FUNC0(SCHEMAinitalize)`

as the FAKE_VOID_FUNC is declared as 
`#define FAKE_VOID_FUNC(...)     EXPAND(FUNC_VOID_(PP_NARG_MINUS1(__VA_ARGS__), __VA_ARGS__))
`

I guess the preprocessor warning applies to a macro definition deeper in fff.h, possibly an unintended consequence of the EXPAND macro?

wulfgarpro, 2018-11-23:
@cshorler, have you tried the latest FFF?  This example works without a problem for me:

```c
#include "fff.h"

DEFINE_FFF_GLOBALS;

void SomeFunc();
FAKE_VOID_FUNC(SomeFunc);

int main(void)
{
    return 0;
}
```

My compiler on LSW:
```bash
$ gcc --version
gcc (Ubuntu 5.4.0-6ubuntu1~16.04.10) 5.4.0 20160609
...
```

cshorler, 2018-11-24:
I think this is explained by a difference in compiler switches (note the warning is for ISO C) - using 
```$ gcc -pedantic -std=c11 -o ftest test.c```
I still see this warning

```
$ gcc --version
gcc (SUSE Linux) 7.3.1 20180323 [gcc-7-branch revision 258812]
```

wulfgarpro, 2018-11-25:
@cshorler, thank you for the followup.  I didn't notice you were trying to compile standard, sorry for the confusion.

I would expect the warning applies to the nested macro:
```c
#define FUNC_VOID_(N,...)     EXPAND(FUNC_VOID_N(N,__VA_ARGS__))
```

In this case, there are no varargs for the 0'th void fake, and therefore this warning applies.

Here is a reduced example:
```c
#include <stdio.h>

void myfunc(char* arg, ...) {
    printf("arg: %s\n", arg);
}
#define something(arg, ...) myfunc(arg, ## __VA_ARGS__)

int main(void)
{
    something("test");          // warning
    something("test", 1);       // no warning

    return 0;
}

// gcc -o test test.c -pedantic
```

AFAICT, this isn't a bug, but a limitation in FFF's generic approach, and as you know:
1. There are no other warnings when using -pedantic, indicating FFF is ISO compliant otherwise
2. Directly calling the 0th void fake works without a warning

Therefore, I would suggest either:
1. Just ignore the warning
2. Use an inline pragma for your test if warnings are treated as errors
3. Continue with your approach

In the meantime, I will try and find some time to create a patch.

cshorler, 2018-11-25:
No probs - this is basically what I expected after taking a cursory look at FFF code, perhaps the bug title should include ISO C.

If this approach exists for MSVC & LLVM as well - it could be interesting to see if the it's possible to apply in the fff.h 
https://stackoverflow.com/a/43974887/1162349

The other answer -isystem / GCC system pragma (for the link above), I'm not so sure - need to test if my "real" errors still appear.