meekrosoft/fff #55: Fake function problem

ekeroc, 2018-08-07:
Hi, could I still use the original function after I faked this function in my unit test?

wulfgarpro, 2018-11-23:
@ekeroc, as mentioned, use the language features.  Keep a function pointer and store the original function before assigning the custom delegate.

Feel free to ask for more information or, if you've found a solution, post an example.

closed 2018-11-23 by wulfgarpro

KasperZK, 2021-11-10:
@wulfgarpro could you elaborate please? I am having the same issue. I am trying to fake functionA, which lives in .h file which I have included in my .cpp unit test file. I am trying to fake the function with FAKE_VALUE_FUNC(bool, functionA); but I get the errors "one or more multiply defined symbols found" and "funcationA already defined in .h". What can I do?

Thanks

wulfgarpro, 2021-11-10:
@KasperZK can you provide a cut down example?

KasperZK, 2021-11-10:
@wulfgarpro 
sure

`
#include "pch.h"
#include "CppUnitTest.h"
#include "..\fff.h"

extern "C"
{
	#include "ioDigitalInput.h"
}

DEFINE_FFF_GLOBALS;

FAKE_VALUE_FUNC(bool, ioFunc);

using namespace Microsoft::VisualStudio::CppUnitTestFramework;

namespace UnitTestfff
{
	TEST_CLASS(UnitTestfff)
	{
	public:
		
		TEST_METHOD(TestMethod1)
		{

		}
	};
}
`
ioFunc is implemented in ioDigitalInput.c

KasperZK, 2021-11-11:
It should probably also be said that I am building the production code from another project within the same solution as the test project is placed in VS2019. The library files from the production code are then used in the test project.