meekrosoft/fff #57: How to obtain variadic function arguments past the fixed number of args?

ykoehler, 2018-10-05:
Given a variadic function ABC(a,b,c,...) I can retrieve arg0_val, arg1_val and arg2_val, but past that I 
get no argument present.

Is there a way to capture the last arguments using something like args_val as a va_list or something like that?

rubiot, 2018-10-05:
You can by using a custom fake fuction: For instance:
```c
FAKE_VALUE_FUNC_VARARG(int, fprintf, FILE *, const char*, ...);
```

In order to access the variadic parameters from your custom fake function, declare a va_list parameter. For instance, a custom fake for fprintf() could call the real fprintf() like this:

```c
int fprintf_custom(FILE *stream, const char *format, va_list ap) {
  if (fprintf0_fake.return_val < 0) // should we fail?
    return fprintf0_fake.return_val;
  return vfprintf(stream, format, ap);
}
```

ykoehler, 2018-10-05:
Yes, I realized that afterwards, but I would prefer the other way since I could validate “after” running the function.

rubiot, 2018-10-05:
You're talking about the call history, right? The problem is the variadic arg types are not known, so there's no way to declare an array to automatically store them. But if you can predict the types you can create you're own history using the custom fake.

bertin0, 2025-10-29:
If I want to actually parse some of the args inside the custom function I get an error since the function signature doesn't end in ...:

```
‘va_start’ used in function with fixed args
```

how can I get around this? If I define the function with ... it complains when assigning to .custom_fake