meekrosoft/fff #58: Help - looking for co-maintainer

meekrosoft, 2018-10-06:
Hi,

fff started as a small scratch-your-own-itch project, and I'm really happy to see so many people getting value from it now.

It has been quite some years since I've been working in embedded, and I have to admit I am perhaps not best suited to maintaining the project going forward.  Basically I'm too busy and underqualified to make a call on many of the pull requests.

With that in mind, I'd really rather give the community the opportunity to take over.  

The main concern I have is to ensure that fff doesn't introduce breaking changes, and remains a single header file. 

If you are interested in contributing to the ongoing governance of the project, please drop a comment below.

Thanks!

wulfgarpro, 2018-11-15:
Not sure that I'm any more 'qualified', but I'm happy to provide co maintenance.  Happy to get in contact via other channels of you'd like to discuss.

meekrosoft, 2018-11-22:
Hi @wulfgarpro - fantastic, great that you are interested!  
I will add you as a contributor to the project.  From here on out, you are most welcome to jump into the PR's and issues.

wulfgarpro, 2018-11-22:
Great, thanks @meekrosoft.