meekrosoft/fff #60: Issue faking out variadic printf

CanadianHoser, 2018-10-16:
I'm trying to fake out printf's in my production code, but any statements without arguments are printed out.  Here's an example file:

#include "CppUTest/TestHarness.h"
#include <stdio.h>
#include "fff.h"

DECLARE_FAKE_VALUE_FUNC_VARARG(int, printf, const char *, ...);

int mock_printf(const char * format, va_list arg)
{
    int result = vfprintf(stdout, format, arg);
    fprintf(stdout, "Goodbye\n");
    return(result);
}

TEST_GROUP(printf_mocks)
{
      void setup() {
      }
      void teardown() {
            RESET_FAKE(printf);
      }
};  

TEST(printf_mocks, printf_test)
{
    fprintf(stdout, "PRINTF_MOCKS (Tests 1 & 2):\n");
    printf("%d:",1); // This is not seen (mock works)
    printf("2:\n");  // This hits normal printf()
    printf_fake.custom_fake = mock_printf;
    fprintf(stdout, "FAKE PRINTF INSTALLED (Tests 3 & 4):\n");
    printf("%d:",3); // This hits mock_printf() - See Goodbye
    printf("4:\n");  // This hits normal printf()
}

When run, I see the following:
whein@sjc-ads-6912:UnitTest >$ ./platform_unit_tests_tests -n printf_test 
PRINTF_MOCKS (Tests 1 & 2):
2:
FAKE PRINTF INSTALLED (Tests 3 & 4):
3:: Goodbye
4:
.
OK (23 tests, 1 ran, 0 checks, 0 ignored, 22 filtered out, 0 ms)

Test cases 2 and 4 are not being handled by FFF, instead going to stdio's printf().  Is there a way to handle this?

cormacc, 2018-11-06:
Looks like the VAARG mock requires at least one arg. Have you tried letting the format string be the first arg -- i.e. something like this:
```
DECLARE_FAKE_VALUE_FUNC_VARARG(int, printf, ...);

int mock_printf(va_list arg)
{
    int result = vfprintf(stdout, arg);
    fprintf(stdout, "Goodbye\n");
    return(result);
}
```

wulfgarpro, 2018-11-23:
@CanadianHoser, have you attempted @cormacc's suggestion?

wulfgarpro, 2018-11-28:
Looks like the generator for the VARARG fakes, only generates N=2-20:
```c
 DECLARE_FAKE_VALUE_FUNC2_VARARG(RETURN_TYPE, FUNCNAME, ARG0_TYPE, ...) \
```

We could probably manage N=1-20 for these kinds of situtations:
```c
 DECLARE_FAKE_VALUE_FUNC1_VARARG(RETURN_TYPE, FUNCNAME, ARG0_TYPE) \
```

I'll see if I can put together a patch over the next couple of days.

CanadianHoser, 2018-11-28:
The suggestion by @cormacc won't work because it doesn't match the vfprintf prototype:

printf_test.cpp:12:38: error: too few arguments to function ‘int vfprintf(FILE*, const char*, __gnuc_va_list)’
     int result = vfprintf(stdout, arg);

CanadianHoser, 2018-11-28:
I do think there's something to our system implementation.  The code listed above does perform as expected (Tests 1 and 2 are suppressed, 3 & 4 are displayed) when compiled standalone, but the same code integrated into our much larger code base (which overrides printf elsewhere) behaves as indicated above.  I did try putting put the mock_printf() implementation into a separate .c file to confirm it's not a compiler difference.

wulfgarpro, 2018-11-28:
@CanadianHoser, it looks like the generator for VARARG fakes, only generates N=2-20:
```c
 DECLARE_FAKE_VALUE_FUNC2_VARARG(RETURN_TYPE, FUNCNAME, ARG0_TYPE, ...) \
```

We could probably manage N=1 for these kinds of situtations:
```c
 DECLARE_FAKE_VALUE_FUNC1_VARARG(RETURN_TYPE, FUNCNAME, ARG0_TYPE) \
```

I'll see if I can put together a patch over the next couple of days.

rubiot, 2021-04-14:
It seems you're just declaring the mock, but not defining it. You either have to use `FAKE_VALUE_FUNC_VARARG`, which directly defines your mock function, or use the pair `DEFINE_FAKE_VALUE_FUNC_VARARG`/`DECLARE_FAKE_VALUE_FUNC_VARARG` if you want to split the declaration/definition.

AravindGopala, 2021-11-06:
@rubiot 
I am having a similar issue with fprintf mocking, In the below cide I tried to mock the fprintf from the standard library, unless I provided at least two parameters in the varArg list, the mocked fprintf is not being called, So the SampleTestPass will successfully call the mock fprintf function but the SampleTestFAIL will never call the mock function instead it calls the fprintf from the standard library as only one argument is provided in the varArgList.

can someone suggest what I am doing wrong:

```
#include "unity.h"
#include "unity_fixture.h"
#include "fff.h"
DEFINE_FFF_GLOBALS;

FAKE_VALUE_FUNC_VARARG(int, fprintf, FILE *, const char *, ...);

TEST(SampleTestGroup, SampleTestPass)
{
    FILE * fd = fopen("/dev/stdout", "w");
    fprintf(fd,"%s, %s", "first_var_arg", "second_var_arg");
    printf("Datla-> (%s)", fprintf_fake.arg1_val);
}

TEST(SampleTestGroup, SampleTestFAIL)
{
    FILE * fd = fopen("/dev/stdout", "w");
    fprintf(fd,"%s", "first_var_arg", "second_var_arg");
    printf("Datla-> (%s)", fprintf_fake.arg1_val);
}

TEST_GROUP_RUNNER(GnssCoreIntfTestGroup)
{
    RUN_TEST_CASE(SampleTestGroup, SampleTest1);
    RUN_TEST_CASE(SampleTestGroup, SampleTestFAIL);

}
static void RunAllTests() {
    RUN_TEST_GROUP(SampleTestGroup);
}

/**
 * @brief Main entry point
 */
int main(int argc, char const *argv[])
{
    return UnityMain(argc, argv, RunAllTests);
}

wilson1375, 2024-05-13:
Has there been a solution to this?