meekrosoft/fff #68: fff needs code coverage

wulfgarpro, 2018-12-05:
With the ever increasing PRs and additional features, fff could do with some code coverage statistics.

This is a perfect write up on gcov for TravisCI and OpenCppCoverage for AppVeyor.

https://codingnest.com/how-to-get-code-coverage-from-ci/

meekrosoft, 2018-12-06:
Sounds like a great idea....although I wonder how good the coverage data will be?  If we don't use a pre-processor macro it's code definition will not end up in a translation unit.  So maybe we need to force instantiate the macros somehow?

wulfgarpro, 2018-12-06:
Isn't that what the tests do?

wulfgarpro, 2018-12-11:
Yes, you're correct.  The pre-processor gets in the way of coverage testing... but, here are some interesting stats from just running gcov on fff_test_c:

```bash
$ gcov -b -c fff_test_c
File 'fff_test_c.c'
Lines executed:100.00% of 74
Branches executed:100.00% of 4
Taken at least once:50.00% of 4
Calls executed:98.77% of 163
Creating 'fff_test_c.c.gcov'

File 'test_cases.include'
Lines executed:77.39% of 345
Branches executed:85.61% of 278
Taken at least once:45.32% of 278
Calls executed:31.25% of 208
Creating 'test_cases.include.gcov'
```

wulfgarpro, 2018-12-11:
Yes, you're correct.  The preprocessor gets in the way of coverage testing.  I'll need to rethink this.

meekrosoft, 2018-12-12:
Might still be useful to ensure the tests actually get run though?

wulfgarpro, 2019-02-13:
@meekrosoft, I had a little look at this just now, a cursory idea, haven't thought hard enough about reasons this might not be useful:

```bash
gcc -E fff.h > out1.txt
gcc -E test/fff_test_c.c > out2.txt
awk '[$a]++' out1.txt out2.txt | wc -l  # 2834
cat out1.txt | wc -l  # 327
cat out2.txt | wc -l  # 3898
# total lines: 327+3898=4225
# percentage of match: 2834/4225*100=~67%
```

With this method, we could calculate an average across all test files and use it as a heuristic.

wulfgarpro, 2019-02-16:
Yes, the above method doesn't work given the preprocessed c file, e.g. `test/fff_test_c.c` includes all of `fff.h`.

Another approach that I'd like to test some more:
```bash

# test/fff_test_c.c
gcc -E test/fff_test_c.c > out
comm -13 <(sort < fff.h | uniq) <(sort < out | uniq) > unique_to_out             # strip out fff.h include
awk 'FNR==NR{a[$1];next}($1 in a){print}'  unique_to_out fff.h > matched  # find unique matches
grep -Fxf  <(sort < matched | uniq) <(sort < fff.h | uniq) > coverage             # find coverage of matches in fff.h

fff_len=`cat fff.h | wc -l`
cov_len=`cat coverage | wc -l`
cov1=`$cov_len / $fff_len`

# test/fff_test_cpp.cpp
gcc -E test/fff_test_cpp.cpp > out
comm -13 <(sort < fff.h | uniq) <(sort < out | uniq) > unique_to_out
awk 'FNR==NR{a[$1];next}($1 in a){print}'  unique_to_out fff.h > matched
grep -Fxf  <(sort < matched | uniq) <(sort < fff.h | uniq) > coverage

cov_len_2=`cat coverage | wc -l`
cov2=`$cov_len_2 / $fff_len`

# total coverage
total_cov=`$cov_len1 + $cov_len2 / ($fff_len * 2)`
```

meekrosoft, 2019-02-25:
Mmm, interesting approach....