meekrosoft/fff #69: FUNCNAME##_reset() conflicts with existing functions

davix, 2018-12-17:
**Describe the bug**
I have two functions: foobar_testcli() and foobar_testcli_reset() in my foobar.h

fff will generate below fake functions:
50 DECLARE_FAKE_VALUE_FUNC3(int, foobar_testcli, foo_info_t*, int, int);
 52 DECLARE_FAKE_VOID_FUNC1(foobar_testcli_reset, foo_info_t*);

But DECLARE_FAKE_VALUE_FUNC3() internally has a FUNCNAME##_reset() that conflicts with my existing foobar_reset().

Am I missing some options to bypass this?

meekrosoft, 2018-12-18:
Hi @davix,

Yes, this is unfortunate.  We should prefix internal fff functions with `fff_` or something similar to put our functions in a namespace of sorts.

I think we can fix this pretty easily.  Watch this space.

davix, 2018-12-20:
Thank you @meekrosoft. Any workaround in this case?