meekrosoft/fff #70: Limit scope of mock functions to file

nitishsprabhu, 2019-01-14:
Limiting functions to file scope can help selectively use the mock implementation or the actual implementation in _certain_ cases. 

For example:
The module under test is mod.c and its dependency is libdep.so. I can have test cases in 2 files, test_file1.c and test_file2.c. 
In test_file1.c, I would like to use the mock functions, thus declare and define them statically in the same file. As the dependent functions get resolved before linking in case of test_file1.c, the mock functions will get used. In case of test_file2.c, the dependent functions get resolved to actual implementation (from libdep.so) at link time as mock functions are not visible to this file. 

In this context, will it be useful to have the capability to define mock implementations as static, with a macro like FAKE_STATIC_VOID_FUNC() and FAKE_STATIC_VALUE_FUNC()?

nitishsprabhu, 2019-01-14:
This approach will most likely not compile. Tried it out with some sample functions. 

The 'dependency_function()' mock was declared static with file scope, but the original implementation was not. GCC throws the error "static declaration of dependency_function() follows non-static declaration".

closed 2019-01-14 by nitishsprabhu

wulfgarpro, 2019-01-14:
Thanks for the idea/ticket.

When compared with the full mocking solutions of high(er) level languages,
the preprocessor is always quite limited.