meekrosoft/fff #74: Disable fakes between test suits (use original functions when needed)

RonAmihai, 2019-03-23:
Let's say I've 2 units (header & source) under the same C module - A (which is independent) and B (which is dependent on unit A's functions). 

I want to write unit tests for both unit A and unit B. 
In order to unit test unit B under it's own test suite, I need to define fakes for unit A's functions - but then I cannot use unit A's original functions in unit A's test suit... 

Is there any way I can "disable" fakes between the test suits in order to be able to use the original functions, before/after "mocking" them?

thomasfinchbr, 2019-09-23:
Hi, were able to solve this issue?
I came across the same situation...
Thanks.

NINI1988, 2019-09-23:
I use my makefile to build two test applications. One where A is real and everything else is mocked. And the second where B is real and everything else is mocked.

thomasfinchbr, 2019-09-25:
Thanks for your answer.
I will use the same approach. It solves the issue.

Darwin-Li-001, 2022-06-20:
> 

can you please tell me how you fixed this?
I laways get this problem:
overloaded function with no contextual type information
 FAKE_VOID_FUNC(pendingResetNodesByPowerOff, uint8_t);