meekrosoft/fff #75: Using FAKE_VOID_FUNC on vsyslog cause  a sizeof-array-argument error

ykoehler, 2019-05-09:
**Describe the bug**
FAKE_VALUE_FUNC(vsyslog, int, const char*, va_list);

is causing a sizeof of va_list which gcc turns into an error and therefore trigger the error sizeof-array-argument

**To Reproduce**
Compile the above

**Expected behavior**
detect array type and not call sizeof on them?

**Screenshots**
In file included from /home/ykoehler/fake_log.cpp:18:
/home/ykoehler/fake_log.cpp: In function ‘void vsyslog(int, const char*, __va_list_tag*)’:
/home/ykoehler/fff/include/fff.h:69:79: error: ‘sizeof’ on array function parameter ‘arg2’ will return size of ‘__va_list_tag*’ [-Werror=sizeof-array-argument]
     memcpy((void*)&FUNCNAME##_fake.arg##n##_val, (void*)&arg##n, sizeof(arg##n));
**Compiler, toolset, platform (please complete the following information):**
 - OS: [e.g. Ubuntu 16.04 LTS]
 - Compiler: [e.g. gcc (Ubuntu 5.4.0-6ubuntu1~16.04.10) 5.4.0 20160609]

**Additional context**

ballenthin, 2020-03-26:
I run into the same issues, is there already a work-a-round available?

tastymushroom, 2020-06-10:
Same here. 
gcc (Ubuntu 7.5.0-3ubuntu1~18.04) 7.5.0

Did anyone come up with a solution yet? Thanks.

alukichev, 2022-09-19:
Looks like it's hard to find info on mocking functions that take array arguments. `typedef` array to a user type for them results precisely in the reported error.

jerry90908, 2023-03-30:
Same issue here.
When original function input argument is 1D array, in previous compiler version(7.5.0), I was able to use pointer type to fake it.
However in gcc version 11.3.0, the following error is complained by the compiler, 

> fff.h:1680:64: error: argument 1 of type ‘uint8_t *’ {aka ‘unsigned char *’} declared as a pointer [-Werror=array-parameter=]

gcc version: cc (Ubuntu 11.3.0-1ubuntu1~22.04) 11.3.0

sung-gon-kim, 2025-03-09:
This PR https://github.com/meekrosoft/fff/pull/139 might be able to resolve the issue.