meekrosoft/fff #76: gtest conflicting

nrbrook, 2019-06-18:
**Describe the bug**
Because gtest is embedded in the root of the respository under the gtest directory, if this repository is included in a unit testing `CMakeLists.txt` using `include_directories`, and googletest is also used in the same project then `#include "gtest/gtest.h"` could include either header. As the gtest.h in this project is very old, this causes problems.

**To Reproduce**
Steps to reproduce the behavior:
1. Create a unit testing project
1. Include this repository directory
1. Also add googletest
1. From a unit test, include `gtest/gtest.h`

meekrosoft, 2019-07-03:
Yes, I can see this could cause a problem.  What do you suggest we do to solve this?

nrbrook, 2019-07-03:
The simple solution would be just to rename the directory, perhaps to `lib/gtest`. In my projects I use CMake which can download googletest during configuration. Alternatively googletest could be added as a git submodule. As the googletest used here is quite old there may be compatibility issues when updating.

eaibmz, 2021-09-16:
This is how i solved this in my CMake file:

```
FetchContent_Declare(fff
    GIT_REPOSITORY https://github.com/meekrosoft/fff.git
    GIT_TAG v1.1)
FetchContent_GetProperties(fff)
if(NOT fff_POPULATED)
    FetchContent_Populate(fff)
    file(REMOVE ${fff_SOURCE_DIR}/gtest)
endif()

```

ztnel, 2022-02-05:
+1 to this