meekrosoft/fff #91: FFF Errors out with array parameters

darrylpinto, 2020-06-04:
**Describe the bug**
Hi, I recently started using FFF and it has worked wonders for me. 
While faking a function with array parameters I noticed a bug. In the following code, I am trying to fake test123() that is being used in test456().
test123() has an array parameter
```
int test123(int a[], int len) {
    if (len == 0) {
        return -1;
    }

    return a[0];
}

int test456() {
    int a[5] = {10,20,30,40,50};
    return test123(a, 5);
}
```

Here is the FAKE_VALUE_FUNC macro for test123 and the testcase:
```
#include <gtest/gtest.h>
#include "fff.h"
DEFINE_FFF_GLOBALS;

FAKE_VALUE_FUNC(int, test123, int [], int);
class TestEg : public ::testing::Test {
public:
    static void SetUpTestSuite() {
    }

    static void TearDownTestSuite() {
    }

    void SetUp() {
        RESET_FAKE(test123);
        FFF_RESET_HISTORY();
    }

    void TearDown() {
    }
};

TEST_F(TestEg, test456) {
    test123_fake.return_val = 100;
    EXPECT_EQ(1, test123_fake.call_count);
    EXPECT_EQ(100, test456());
    EXPECT_EQ(5, test123_fake.arg1_history[0]);
}
```

On running, this is what I am getting:
```
 error: expected unqualified-id before ‘[’ token
 error: expected unqualified-id before ‘[’ token
 error: expected ‘,’ or ‘...’ before ‘arg0’
 error: expected ‘,’ or ‘...’ before ‘arg0’
 error: expected ‘,’ or ‘...’ before ‘arg0’
 error: expected ‘,’ or ‘...’ before ‘arg0’
In function ‘int test123(int*)’:
 error: ‘test123_Fake’ has no member named ‘arg0_val’
 error: ‘arg0’ was not declared in this scope
 error: ‘arg1’ was not declared in this scope
 error: ‘test123_Fake’ has no member named ‘arg0_history’
 error: overloaded function with no contextual type information
```

I change the signature to use pointer instead of array:
`FAKE_VALUE_FUNC(int, test123, int *, int);`

Now on running, I get the following, I don't get errors, but faking of test123 fails:
```
[==========] Running 1 test from 1 test suite.
[----------] Global test environment set-up.
[----------] 1 test from TestEg
[ RUN      ] TestEg.test456
testeg_gtest.cpp:68: Failure
Expected equality of these values:
  1
  test123_fake.call_count
    Which is: 0
[  FAILED  ] TestEg.test456 (0 ms)
[----------] 1 test from TestEg (0 ms total)

[----------] Global test environment tear-down
[==========] 1 test from 1 test suite ran. (0 ms total)
[  PASSED  ] 0 tests.
[  FAILED  ] 1 test, listed below:
[  FAILED  ] TestEg.test456

```

Surprisingly, `EXPECT_EQ(100, test456())` returns True, so that means return_val is working.

Could you please suggest what should I do to fix the original FAKE_VALUE_FUNC macro?

**Expected behavior**
`FAKE_VALUE_FUNC(int, test123, int [], int)` should not give an error

**Compiler, toolset, platform (please complete the following information):**
OS: Ubuntu 18.10
Compiler: gcc (Ubuntu 8.2.0-7ubuntu1)

NINI1988, 2020-06-04:
The test works correctly: test456() has not been called, so test123() has not been called before checking the call_count.

meekrosoft, 2020-06-05:
Can we close this?

darrylpinto, 2020-06-05:
Thank you so much for pointing that out. So the correct way of using FFF with array parameters would be to use the pointer representation instead, right?

closed 2020-06-08 by darrylpinto