meekrosoft/fff #94: Unable to reuse fakes across multiple test modules

samgalactic, 2020-11-20:
I am trying to create Fakes for unit tests in an embedded C project. I can create a single unit test module which will work well, but there are some functions calls which are common to multiple modules which need to be faked in multiple test modules, and this is causing problems. 

For example, I have a Fake function defined as:

FAKE_VOID_FUNC(TraceAssert_Report, uint8_t, uint32_t);

which appears in two separate test modules, and causes the link error:

Error    multiple definition of `TraceAssert_Report_fake' UnitTests [ProjDir]\HeaterTests.o           0   

Error    multiple definition of `TraceAssert_Report_reset()' UnitTests [ProjDir]\HeaterTests.cpp    33  

I have attempted to use to solution presented in the README.md, which is to separate definition and declaration. So I then have a header file with

DECLARE_FAKE_VOID_FUNC(TraceAssert_Report, uint8_t, uint32_t);

and two separate source files which both have 

DEFINE_FAKE_VOID_FUNC(TraceAssert_Report, uint8_t, uint32_t);

but the error messages are exactly the same with this approach.

Is there a way to Fake a function across multiple test modules without hitting multiple definition errors?

Using Windows 10. Visual Studio 2015, Visual GDB, Toolchain arm-eabi

Lizhangz, 2020-11-21:
If you need common behavior, you can just define it in a common c file.
If you need different behavior for different cases, you should use the custom_fake or SET_RETURN_SEQ.

samgalactic, 2020-11-23:
I now have a common .h file where I have the DECLARE_FAKE_   and I have done as you suggested and used a common .cpp file when I have put the DEFINE_FAKE_ .  

There is a problem with this however. If you attempt te

closed 2020-11-23 by samgalactic