meekrosoft/fff #97: Alternating fff with real functions to access the hardware

cakira, 2021-02-16:
Hi!
I'm still a newbie in unit tests, so I'm sorry if I have a wrong concept here.

I'm working in an embedded system and I designed a unity test for a module that transfers data from an incoming queue to a Flash memory and from the Flash memory to an outgoing queue. In this unity test I'm using fff to mock the Flash memory functions. One test is more or less like this:
```c
void test_store_one_ip_address() {
    const char my_ip[] = "123.45.67.89";
    queue_send_to_back(&ip_queue, my_ip, sizeof(my_ip));

    nvs_manager_loop();
    TEST_ASSERT_EQUAL_INT(1, write_flash_fake.call_count);
    TEST_ASSERT_EQUAL_STRING("ip0", write_flash_fake.arg1_val);
    TEST_ASSERT_EQUAL_STRING(my_ip, write_flash_fake.arg2_val);
}
```
But now I would like to run some of the test cases using use the real Flash memory functions. My motivation to run the tests with real hardware is that sometimes the hardware and/or the documentation can be treacherous: once I had thought that I understood the API to use a DMA, but I didn't, therefore my module worked perfectly with the mock, but not with the hardware.

The problem is that when I use the real functions, there is no more `write_flash_fake`. I had to comment out the lines to compile it and then, by other means, check that the memory was indeed written.

I was thinking about using `custom_fake` and some gimmicks with the preprocessor (like `#ifdef HARDWARE` `#define write_flash real_write_flash` ... `write_flash_fake.custom_fake = real_write_flash` or something like that) but I couldn't find one that made the test above to work. Is it possible?

Does this concept even make sense?

Thank you!

NINI1988, 2021-02-16:
What I do is to run unit test for eg. the queue on my pc and create another application to test only the flash on the hardware. When both tests are well written the queue and the flash will work together on the hardware.

cakira, 2021-02-18:
Now that you say it, it makes perfect sense.
I'll do that way too.

Thank you very much, @NINI1988 !

closed 2021-02-18 by cakira