meekrosoft/fff #106: Is it possible to mock static functions?

KasperZK, 2021-11-22:
I presented the framework to my team and it raised some questions:

Can you mock static functions with this framework? For example would it be possible to mock and test a static function that is both defined and declared in the .c file that is to be tested? Do we even want to do this ever or should private functions only be accessed through public functions?

Thanks

nradulovic, 2021-11-22:
Hi, you can only mock functions which are defined outside of tested implementation unit (.c). 

In my projects I treat static functions as part of tested function so I don't need to mock it. When I get into the situation that I need to mock a static function, then that is a sign to me that I need to re-think about architecture of the code and ask some questions to myself.

Puncher1, 2024-10-23:
I usually use static functions, if it's only used "internally" i.e. in the .c file itself and doesn't have to be included somewhere else. In that case I don't think it's a code architecture problem, but how would you mock that static function.

For example, I have a function called `I2cTransmit` which is "public" and can be called externally. This function calls the static function `I2cStartCondition` which returns a value. In this example, how would I mock `I2cStartCondition` to "fake" the return value to then test the behaviour of  `I2cTransmit`?

> In my projects I treat static functions as part of tested function so I don't need to mock them.

That makes sense, especially if you want to test what the static function itself does, but you can't test what the behaviour of the function is that calls the static function, or am I missing something?