meekrosoft/fff #118: fake.callcount is always 0 when reuse a fake across multiple test-suites

Darwin-Li-001, 2022-11-13:
**Describe the bug**
A clear and concise description of what the bug is.
When reuse a fake across multiple test-suites, Functions with parameters faked as the instruction described, but the callcount is always 0, however the functions without parameters callcount is normal.
**To Reproduce**
Steps to reproduce the behavior:
1.Fake a function with parameters as instruction described, with DECLARE_FAKE_VOID_FUNC in a public .h file;
2. use DEFINE_FAKE_VOID_FUNC in a private .c file
3. Reference faked function in a Google Test test suit TEST_F;
4. When I try to use the FUN_fake.callcount, it is always 0; 
5. but if i fake the function directly in the test source file, the callcount is OK.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
[image]

**Compiler, toolset, platform (please complete the following information):**
gcc.exe (x86_64-posix-seh-rev0, Built by MinGW-W64 project) 8.1.0
vscode V1.72

Darwin-Li-001, 2022-11-16:
The problem is still nobody support now.

Darwin-Li-001, 2022-11-17:
@meekrosoft @michahoiting @zrax @wulfgarpro   Anybody here can help this? This is urgent.  Thanks!

ff12323, 2022-12-18:
use g++ -E , u will see its a simple Preprocessing Seams, just define a function the same name, then link call it.

the reset just memset value 0. Cannot change called func

```
void checkSkipList_reset(void)
{
  memset((void *)&checkSkipList_fake, 0, sizeof(checkSkipList_fake) - sizeof(checkSkipList_fake.custom_fake) - sizeof(checkSkipList_fake.custom_fake_seq));
  checkSkipList_fake.custom_fake =
# 6 ".\\fff_test.cpp" 3 4
      __null
# 6 ".\\fff_test.cpp"
      ;
  checkSkipList_fake.custom_fake_seq =
# 6 ".\\fff_test.cpp" 3 4
      __null
# 6 ".\\fff_test.cpp"
      ;
  checkSkipList_fake.arg_history_len = (50u);
};
```

Darwin-Li-001, 2022-12-19:
> et value 0. Cannot change called func

Thank you for your reply!

Can we fix it since you know the root cause?