meekrosoft/fff #122: FFF generated reset functions have mangled names

sidwarkd, 2023-01-26:
**Describe the bug**
I have an embedded project and am trying to follow the pattern in the weak linking example. I'm able to get my tests to run with FFF fakes including things like custom_fake, call_count, etc but I can't reset the fakes. Whenever I add a call to `RESET_FAKE` my test program fails to link with the errors:

```txt

...undefined reference to `esp_http_client_perform_reset'
...undefined reference to `esp_http_client_init_reset'
...undefined reference to `esp_http_client_perform_reset'
...undefined reference to `esp_http_client_init_reset'
...undefined reference to `esp_http_client_perform_reset'
...undefined reference to `esp_http_client_init_reset'
```
When I run `nm` on one of the object files that goes into my libfakes library I see the following output:

```txt
                 U _GLOBAL_OFFSET_TABLE_
0000000000001867 T _Z21esp_err_to_name_resetv
0000000000000331 T _Z26esp_http_client_init_resetv
0000000000001ade T _Z29_esp_error_check_failed_resetv
00000000000006a2 T _Z29esp_http_client_cleanup_resetv
0000000000000a13 T _Z29esp_http_client_perform_resetv
0000000000000e22 T _Z32esp_http_client_set_header_resetv
0000000000001504 T _Z37esp_http_client_get_status_code_resetv
0000000000001193 T _Z40esp_http_client_get_content_length_resetv
0000000000001e2d T _Z41esp_http_client_is_chunked_response_resetv
00000000000018a4 W _esp_error_check_failed
0000000000001680 B _esp_error_check_failed_fake
0000000000001541 W esp_err_to_name
00000000000013e0 B esp_err_to_name_fake
000000000000036e W esp_http_client_cleanup
0000000000000380 B esp_http_client_cleanup_fake
0000000000000e5f W esp_http_client_get_content_length
0000000000000ea0 B esp_http_client_get_content_length_fake
00000000000011d0 W esp_http_client_get_status_code
0000000000001140 B esp_http_client_get_status_code_fake
0000000000000000 W esp_http_client_init
0000000000000000 B esp_http_client_init_fake
0000000000001b1b W esp_http_client_is_chunked_response
0000000000001d20 B esp_http_client_is_chunked_response_fake
00000000000006df W esp_http_client_perform
0000000000000620 B esp_http_client_perform_fake
0000000000000a50 W esp_http_client_set_header
00000000000008c0 B esp_http_client_set_header_fake
                 U fff
                 U memset
```

As you can see, most of the FFF generated functions have clean names but all of the reset functions are mangled. Is this an issue with FFF or how I am generating my libfakes library in CMake. Due to the embedded nature of the project there is a mix of C and C++ code involved.

**Compiler, toolset, platform (please complete the following information):**
 - OS: Docker image running Debian
 - Compiler: gcc (Debian 10.2.1-6) 10.2.1 20210110

paleozogt, 2023-10-10:
This is happening because of C++ vs C linkage.  You can wrap your fake definitions in `extern "C" { ... }`.

sidwarkd, 2023-10-27:
@paleozogt Unfortunately I can't easily get back into a state to test this but it feels like it will be the correct answer. I didn't even think about extern "C"-ing things. I'm going to close the issue and if I get back to this state and that doesn't resolve it I will re-open. Thanks for your help.

closed 2023-10-27 by sidwarkd