meekrosoft/fff #128: Mocking static methods of the classes

jkrajniak, 2023-09-19:
Hi,

Is there any way to mock static methods of the class. I've tried but it fails. Basically
```
FAKE_VOID_FUNC(BaselineCalib::setModel)
```
where setModel is a static method of BaselineCalib class. Now it returns with the error

```
tests/fff.h:154:36: error: qualified name does not name a class before '{' token
  154 |     typedef struct FUNCNAME##_Fake { \
```

jakub-dudarewicz, 2023-09-28:
Hi, just passing by. FFF is supposed to be used for C, not C++. For C++ better use GoogleTest, CppUTest, etc.

jkrajniak, 2023-09-28:
True, although I have a project that mixes C and C++ (with extern C) and then the standard mocking is almost impossible...

jakub-dudarewicz, 2023-09-28:
I haven't yet been in that situation, I guess using FFF for C and Google Mock for C++ is out of the question?